ArduPilot/ardupilot_wiki #2731: Move MAVProxy documentation to wiki

stephendade, 2020-05-09:
Currently, the MAVProxy documentation is over at http://ardupilot.github.io/MAVProxy/html/index.html, with the source code at https://github.com/ArduPilot/MAVProxy/tree/gh-pages

To make things easier for users and maintainers I'm thinking of moving the MAVProxy documentation to the wiki, much like how Mission Planner does.

So there'd be a "new" site in the wiki: https://ardupilot.org/mavproxy/index.html and associated top-level folder in the repo.

Fortunately the MAVProxy documentation is all in rst format so conversion won't be a major issue. I am assuming there'll have to be a configuration change somewhere to include MAVProxy in the wiki build.

I'd be interested in @tridge, @peterbarker and @Hwurzburg's thoughts on this.

[I'm happy to do the work on this]

peterbarker, 2020-05-09:
On Fri, 8 May 2020, Stephen Dade wrote:

> I'd be interested in @tridge, @peterbarker and @Hwurzburg's thoughts on this.

I really like the idea!

> [I'm happy to do the work on this]

Even better! :-)

Hwurzburg, 2020-05-09:
Sounds great...folder needs to be added in ardupilot-wiki with the same structure as other sites like plane, copter, etc....    update.py would need a change....   and the site added in the ardupilot site's index.rst toc at the bottom...

closed 2020-06-13 by stephendade