ArduPilot/ardupilot_wiki #2759: Debugging_arducopter

RakeshVangaveti, 2020-05-17:
hello,
i'm new to ardupilot ,i gone through ardupilot debug documentation to debug it
[Debugging](https://ardupilot.org/dev/docs/debugging.html)
i followed the same and my problem is if i want to debug mode_auto.cpp/mode_etc.cpp any of one files , i put a breakpoint in one of the file some random lines bun can't reach there and debug windows goes to blocking mode
how can i reach that break point ? ,what is the reason ?,can we debug entire arducopter code?
if any one has tried to debug entire arducopter code debug while running sitl in ubuntu..plase give me inputs.(how can debug it without blocking)

pardon me if i ask anything wrong..!
thanks in advance..
hope will get quick reply..

Hwurzburg, 2020-05-17:
try moving this question over to https://gitter.im/ArduPilot/ardupilot for developer questions....thanks!

this is only for documentation issues....

closed 2020-05-17 by Hwurzburg

RakeshVangaveti, 2020-05-17:
thanks @Hwurzburg ...will discuss it there.

RakeshVangaveti, 2020-05-22:
> hello,
> i'm new to ardupilot ,i gone through ardupilot debug documentation to debug it
> https://ardupilot.org/dev/docs/debugging-with-gdb-on-linux.html
> i followed the same and my problem is **while debug mode_auto.cpp/mode_etc.cpp any of one files put a breakpoint** in one of the file some random lines **but can't reach there** and debug window goes to blocking mode
> **SITL (terminal 1)**
> $ cd ardupilot/ArduCopter/
> 
> $ python ../Tools/autotest/sim_vehicle.py -v ArduCopter -f quad --console --map -D -G
> 
> **DEBUGGING(terminal 2)**
> $gdb --args rakesh/ardupilot/build/sitl/bin/arducopter -S -I0 --home -35.363261,149.165230,584,353 --model + --speedup 1 --defaults rakesh/ardupilot/Tools/autotest/default_params/copter.parm
> 
> $(gdb)b mode_auto.cpp:252 //break point at 252 line
> 
> $(gdb) r //run the debugging
> --- //the cursor is blinking contineously(blocking) and sitl window works fine..
> 
> how can i reach that break point ? ,what is the reason ?,can we **debug entire arducopter** code?
> if any one has tried to debug entire arducopter code debug while running sitl in ubuntu..plase give me inputs.(how can debug it without blocking)
> and also **--console --map are not working** (mavproxy windows are not opening) **while running sitl**
> 
> pardon me if i ask anything wrong..!
> thanks in advance..
> hope will get quick reply..

issues solved ,thanks for developer discussion link ..! @Hwurzburg