ArduPilot/ardupilot_wiki #2773: ArduPPM Broken Github Link

wuoliver, 2020-05-22:
The github link to the ArduPPM encoder repo seems to be broken. I've been trying to figure out where it has been moved to, but it seems to be missing entirely. 

Under: 
docs/common-ppm-encoder.rst

Broken Link: 
https://github.com/ArduPilot/ardupilot/tree/master/Tools/ArduPPM

Hwurzburg, 2020-05-22:
https://download.ardupilot.org/downloads/wiki/advanced_user_tools/ArduPPM_v2.3.16_ATMega32U2.hex_.zip

wuoliver, 2020-05-22:
Thanks Henry, but I'm actually looking for the source code, not the binary.

Hwurzburg, 2020-05-22:
@peterbarket  do you know where it is now?

peterbarker, 2020-05-23:
It was removed years ago.

```
pbarker@bluebottle:~/rc/ardupilot((no branch, bisect started on master))$ git bisect good
2c1be910d8279ba6e7e631d510992a4fb2a5e170 is the first bad commit
commit 2c1be910d8279ba6e7e631d510992a4fb2a5e170
Author: Andrew Tridgell <andrew@tridgell.net>
Date:   Thu Jul 5 13:58:45 2018 +1000

    Tools: removed a lot of ancient AVR code

:040000 040000 0e7e7e95f85b9cab18e1bb474dde3bba41a9d6ec 51fc4dda22771498c64f8b567ab440e862c0f697 M	Tools
pbarker@bluebottle:~/rc/ardupilot((no branch, bisect started on master))$ 
```

To get at it, one would probably clone the ardupilot repo, then check `2c1be910d8279ba6e7e631d510992a4fb2a5e170^` out.

wuoliver, 2020-05-28:
Thanks Peter, I found it. 

Does this mean that the binary file linked on the PPM Encoder Wiki page will be the last version released since the code has been removed from the repo?

peterbarker, 2020-05-29:
On Thu, 28 May 2020, Oliver Wu wrote:

> Does this mean that the binary file linked on the PPM Encoder Wiki page will be the last version released since the code has been removed from the repo?

I would guess so.  Curious to know what your use-case now-adays is!

Peter