ArduPilot/ardupilot_wiki #2778: Maxbotix Analog Sonar

AndreKotze, 2020-05-23:
[Maxbotix Analog Sonar](https://ardupilot.org/copter/docs/common-rangefinder-maxbotix-analog.html)

A notice needs to be added to this page to state that whenever changes have been made to the range finder settings in Mission Planner the FC needs to be rebooted before they will take affect.

Hwurzburg, 2020-06-08:
which settings?, besides type....changing peripheral types always requires a reboot since drivers need to be loaded and initialized....most drivers once active accept  run time changes to scales, etc.

AndreKotze, 2020-06-09:
Mainly the type but if I recall correctly the pin also.

Hwurzburg, 2020-06-25:
added note on rangefinder setup page that type change requires reboot to load the driver

closed 2020-06-25 by Hwurzburg