ArduPilot/ardupilot_wiki #2782: Using SITL with Morse

auturgy, 2020-05-25:
Fresh Morse install on Ubuntu 20.04 not working.  This isn't an ArduPilot issue, but it does need to be noted until resolved.  Issue raised on Morse here: https://github.com/morse-simulator/morse/issues/822
Steps to reproduce (from /ardupilot dir):
```sudo apt install morse-simulator
morse run libraries/SITL/examples/Morse/quadcopter.py```
Full console dump is at shorturl.at/svJM6
but basically Blender quits due to uncaught exception, 
AttributeError: 'Object' object has no attribute 'hide'.  
Doesn't seem to be related to ArduPilot - same error occurs regardless if starting the default Morse vehicle.
```morse check``` returns 
...
* Your environment is correctly setup to run MORSE.

```morse create my_first_sim
   morse run my_first_sim ```
returns same error (AttributeError: 'Object' object has no attribute 'hide'.) as running the ardupilot script

peterbarker, 2020-08-17:
I'll also note that if you do hack that attribute "hide" problem (the fix is googleable) then a more intractable exception crops up next.

Workaround *might* be to install Blender 2.79 somehow using snap - theory only.

auturgy, 2022-08-07:
Closed https://github.com/ArduPilot/ardupilot_wiki/pull/3465

closed 2022-08-07 by auturgy