ArduPilot/ardupilot_wiki #2787: Communicating with Raspberry Pi via MAVLink - apt-get failure.

srosro, 2020-05-28:
Article: [Communicating with Raspberry Pi via MAVLink](https://ardupilot.org/dev/docs/raspberry-pi-via-mavlink.html)

The line: `sudo apt-get install python3-dev python3-opencv python3-wxgtk3.0 libxml2-dev python3-pip python3-matplotlib python3-lxml`

appears to instruct the user to install a package (python3-wxgtk3.0) that doesn't exist, whereas python3-wxgtk4.0 does exist:

```
$ apt-cache search python3-wxgtk3.0
$ apt-cache search python3-wxgtk4.0
python3-wxgtk4.0 - Python 3 interface to the wxWidgets Cross-platform C++ GUI toolkit```

srosro, 2020-05-28:
Also related, I don't believe the command `python3 mavproxy.py --master=/dev/serial0 --baudrate 921600 --aircraft MyCopter` would work if mavproxy is installed via pip...you might consider including instructions to `cd` to the correct location first.

Hwurzburg, 2020-05-31:
@stephendade can you comment on this and what if any changes are needed?

auturgy, 2020-05-31:
By following the existing instructions and adding mavproxy to the system PATH you can run it from anywhere. The wxgtk version change I fixed in the mavproxy docs a week or two ago. Might be best to remove the duplicate here and just reference back to the canonical mavproxy instructions.

stephendade, 2020-05-31:
Yep, agree with @auturgy. I think there's a few doubled-up instructions for MAVProxy in the wiki that can now just be references back.

In regards to ``pip install mavproxy --user``, as long as the user has ``$HOME/.local/bin`` in their PATH they can execute MAProxy from anywhere.

auturgy, 2020-05-31:
https://github.com/ArduPilot/MAVProxy/pull/764/files is the change I made.  Unfortunately it depends on which Debian / Ubuntu / Raspbian version the user has, so no universal solution at present.

auturgy, 2020-05-31:
@srosro thanks for reporting this by the way: I often forget that some instructions are repeated in different places.

Hwurzburg, 2020-06-01:
> Yep, agree with @auturgy. I think there's a few doubled-up instructions for MAVProxy in the wiki that can now just be references back.
> 
> In regards to `pip install mavproxy --user`, as long as the user has `$HOME/.local/bin` in their PATH they can execute MAProxy from anywhere.

.. if we just point there, they had better be complete and easy to use...the current install instructions should have code blocks to copy for the various cases shown, not just install "xyz" instead.and also include the differences for the RPi install...

closed 2020-06-21 by Hwurzburg