ArduPilot/ardupilot_wiki #2820: Copter: 4.0.4 updates required

rmackay9, 2020-06-05:
Here is a list of a few items we should add to the wiki ahead of the Copter-4.0.4 release

- Neo and Sui smart battery pages
- GPS for yaw (aka Moving baseline)
- new autopilots

MadRC, 2020-06-05:
I have started putting together a Nora page on my fork, I need to double check a few things but will summit once done 

https://github.com/MadRC/ardupilot_wiki/blob/master/common/source/docs/common-cuav-nora-overview.rst

MadRC, 2020-06-10:
I have built the X7 page as well, I just need to do a local build and check them and then will put in the pull request for both 

https://github.com/MadRC/ardupilot_wiki/blob/master/common/source/docs/common-cuav-X7-overview.rst

Hwurzburg, 2020-06-25:
@rmackay9 Randy, will need help on smart battery stuff, unless you are doing that...
you have another issue on 4.0.4, could you consolidate it here? thanks

SUI-Eric, 2020-06-26:
Hi @Hwurzburg, I'd be happy to help put together pages for both the SUI and Neo battery drivers. I've not contributed to the Wiki before - is there any specific format or information that I should be sure to include?

rmackay9, 2020-06-26:
@SUI-Eric, it should work a bit like this: https://ardupilot.org/copter/docs/common-smart-battery.html

So the bits we probably need are:

- a title image of the battery
- an diagram showing how it can be connected to an autopilot
- a list of required parameter changes.  The link above shows a picture of setting up using the MP but I think recently we just directly mention the parameter changes required

SUI-Eric, 2020-06-26:
Sounds good, I'll get going on that and will reach out if I've got any questions.

Hwurzburg, 2020-10-21:
status? dont recall seeing anything on these batteries...at least not on that page

SUI-Eric, 2020-10-22:
Hi @Hwurzburg - really sorry about this. I haven't forgotten about it but keep getting swamped and it often falls off my radar. I'll see if I can enlist some help on my end to finish this up.