ArduPilot/ardupilot_wiki #2822: Links for old parameters files.

brunoolivieri, 2020-06-05:
The script that makes multiple parameters files enables links for each parameter in each file in an individual a manner. 

It makes the wiki build slower and uses more memory due to the huge toctree (thousands of (un)used links).

IFF we do not link old parameters would be nice remove such links and enable linking only for the latest firmware of each vehicle.

What do you think?

Hwurzburg, 2020-06-05:
we dont link any old rev params I beleive, if they are not in master, they would throw errors, if that happens, I unlink them (on archived pages,etc.) so I dont think links to other than the master set are required...

Hwurzburg, 2020-10-21:
any progress Bruno?is this still an issue?

brunoolivieri, 2020-10-28:
I guess we could close it now.

closed 2020-10-28 by brunoolivieri