ArduPilot/ardupilot_wiki #2841: Misleading Description of Parameter THR_MAX in Parameter List

Rolf-G, 2020-06-11:
The description of [THR_MAX](https://ardupilot.org/plane/docs/parameters.html#thr-max-maximum-throttlel) is: 

"Maximum throttle percentage used in **automatic throttle** modes."

After that you have to assume that THR_MAX is not used in FBWA, because FBWA is not one of the automatic throttle modes . That was annoying when there wasn't enough power to throw the plane in FBWA. (I've found another way to use THR_MAX only in auto throttle modes by enabling THR_PASS_STAB). But either you should adapt the description to the functionality or vice versa.

vierfuffzig, 2020-06-11:
this likely just slipped through when THR_MIN was in focus last year, see https://github.com/ArduPilot/ardupilot/pull/11059 for reference.

vierfuffzig, 2020-06-11:
https://github.com/ArduPilot/ardupilot/pull/14586

vierfuffzig, 2020-06-12:
fix merged, can be closed

rmackay9, 2020-06-12:
Txs!

closed 2020-06-12 by rmackay9