ArduPilot/ardupilot_wiki #2883: SITL Simulator (Software in the Loop)

albcem, 2020-06-29:
[SITL Simulator (Software in the Loop)](https://ardupilot.org/dev/docs/sitl-simulator-software-in-the-loop.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

The figure for the communication architecture for SITL is outdated; the Port numbers and the pairs of software that communicate are wrong. I found the following figure online that correctly represents the current state of the communications. The owner of the figure proposed it to update the affected page but no change has been made.

[image]

brunoolivieri, 2020-06-29:
Link to PR #1764

peterbarker, 2020-08-24:
@Hwurzburg it does address my comments.

Hwurzburg, 2020-08-24:
closed by #2983

closed 2020-08-24 by Hwurzburg