ArduPilot/ardupilot_wiki #2889: Typo for SERIAL5 on MATEK F405-STD page

tamboekie, 2020-07-01:
There is a minor issue with common/source/docs/common-matekf405.rst

The SERIAL5 parameter has a typo and is written as `SEIRAL5`.

It is also not exactly clear which SERIALx parameters correspond to which pads on the board. I have worked that out through trial and error, but perhaps others will benefit it is explictely listed on the page.

tamboekie, 2020-07-01:
I've create a PR #2890

tamboekie, 2020-07-01:
The PR #2890 takes care of the typo, but perhaps somebody could comment on the mapping from SERIALx to board pads: "It is also not exactly clear which SERIALx parameters correspond to which pads on the board. I have worked that out through trial and error, but perhaps others will benefit it is explictely listed on the page."

Maybe there is a more obvious way to work it out?

Hwurzburg, 2020-07-01:
I think its pretty clear on the board silk screen....UARTpads are marked like TX5 and RX1, for UART5 TX and UART1 RX...then the docs show how they are mapped to SERIAL port param

tamboekie, 2020-07-02:
@Hwurzburg, I must be making a mistake.

I have found that RX1/TX1 maps to SERIAL3 and RX3/TX3 maps to SERIAL2.
Then TX2/RX2 maps to SERIAL2, as expected.
Right now I can't get SERIAL4 to work on TX4/RX4, so I have started to wonder if it may be swapped with RX5/TX5.

I think one option is to go to `libraries/AP_HAL_ChibiOS/hwdef/MatekF405/hwdef.dat` and follow the pad mappings from

`# order of UARTs (and USB)`
`SERIAL_ORDER OTG1 USART3 UART4 USART1 UART5 USART2

# USART3
PC10 USART3_TX USART3
PC11 USART3_RX USART3`

...etc.

But that involves ripping my board out from its nice mounting which I would rather not do :)

Hwurzburg, 2020-07-02:
this is the mapping in the hwdef AND docs:

    SERIAL0 = console = USB
    SERIAL1 = Telemetry1 = USART3
    SERIAL2 = Telemetry2 = UART4
    SERIAL3 = GPS1 = USART1
    SERIAL4 = GPS2 = UART5
    SERIAL5 = User = USART2 (TX only unless BRD_ALT_CONFIG = 1, then RX is available)

TX/RX 2 maps to SERIAL5 not SERIAL 2...thats is TX/RX4   SERIAL5 is UART2.... SERIAL4 is UART5....

Hwurzburg, 2020-07-06:
if you are clear now...lets close this

tamboekie, 2020-07-07:
@Hwurzburg, yes I agree. I have taken a hard look at my setup and realised I have misunderstood it all. I think it is clear now. Thanks for helping me out.

closed 2020-07-07 by tamboekie