ArduPilot/ardupilot_wiki #2900: Flight Modes

brunoolivieri, 2020-07-04:
[Flight Modes](https://ardupilot.org/copter/docs/flight-modes.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

hi! :)

So far as I know, in Copter, all flight modes that hold altitude uses the barometer. (Or more complex hw such as a sonar, etc...) 

Take into consideration the Plane: It uses the barometer in the same way? I mean, a simple board in FBWB or Cruise, etc?

Shaw we include a collum with barometer necessity as GPS is presented?

Hwurzburg, 2020-07-05:
until a very recent change you could not even boot any board if you did not have a baro in the system...however, there is now a build option to allow that for a few boards without baro like the speedybee....

its a bit more than just using baro, baro and gps and imu factor into altitude estimation which the altitude holding modes use in copter and plane...

I am not really sure if and how the mode is denied if someone was using a speedybee or MatekF405 std and tried to enter an altitude hold mode....and it may depend on other params...will talk to Randy and Tridge about it

vierfuffzig, 2020-07-05:
i’ll chime in here if i may as i was involved in no-baro handling. you can boot without a baro based on a compile time define that was added to support boards like speedybeef. this was temporarily overriden by a runtime check in baro cal during init, refusing to fully run without a baro present. that‘s what i intended to fix with https://github.com/ArduPilot/ardupilot/pull/14317 as it might be required to adjust some settings for this individual use case, like specify an alternative alt source.
however, you can‘t arm without a valid alt source set and present. that doesn‘t have to be baro necessarily. i‘m not aware of any additional mode check explicitely asking for a baro. happy to do some more testing though.

brunoolivieri, 2020-07-05:
Tks!

@Hwurzburg so, the barometer is just yet another entry for EKF?!  That is fine for me, I guess that was a user doubt from my part because I do not use Plane as much as Copter.

@vierfuffzig I do not think we have anything wrong here. Maybe a reinforcement in the basic setup documentation to protect the barometer from wind for Planes as in Copters XOR check non-baro documentation.

Hwurzburg, 2020-07-05:
@vierfuffzig so what alternate alt sources can be used for baro-less uses? gps? rangefinder? are no modes denied with these alternate sources? what params need to be set?
there seems to be a lot of documenation needed here....but I need to understand all the issues to do  it....

vierfuffzig, 2020-07-05:
imho for plane GPS will likely be the most common alternative, and that allows to use all modes. this requires to set EKn_ALT_SOURCE = 2 (GPS). while current high end hardware will likely beat the baro in terms of precision and reliability, it obviously leaves you without fallback if your GPS fails and you have no baro attached.
rangefinder and beacon are other alt source options, but afaik can only be used additionally alongside other position sensors, just like baro. i really don't know if positioning systems other than GPS are relevant for plane use

vierfuffzig, 2020-07-06:
i've re-read the conversation on https://github.com/ArduPilot/ardupilot/pull/14317 again and did some more testing. imho most importantly baro pre-arm check picks up a missing baro, so there's no way arming unless unticking that check when there's no baro  connected.

peterbarker, 2020-07-06:
On Sun, 5 Jul 2020, vierfuffzig wrote:

> i've re-read the conversation on ArduPilot/ardupilot#14317 again and did some more testing. imho most importantly baro pre-arm check picks up a missing
> baro, so there's no way arming unless unticking that check when there's no baro connected.

This should never be about having a baro.

This should be about having an altitude measurement.

Hwurzburg, 2020-07-06:
well, I dd a little more research...the only two non gps module, hwdefs for autopilots are the Speedybee, which already has a unverified label on it since I could never get an external baro attached (tried both types its supposed to scan for)...and the SuccexxF4, which has no wiki page and doesnt pin out an external bus to attach one anyway...so my concerns about someone taking an approved flight controller and flying it are mute at this point...

Hwurzburg, 2020-07-06:
okay so lets close this one...

closed 2020-07-09 by Hwurzburg