ArduPilot/ardupilot_wiki #2907: Wiring diagram for Matek F405-STD

tamboekie, 2020-07-07:
I have created a comprehensive wiring diagram based around the Matek F405-STD controller for my own use. I wondered if this might be useful for the wiki? My own diagram is quite a mess, but I could clean it up if the maintainers here think it is worthwhile.

[image]

PS: I was not sure if there is a better place to propose such things?

Hwurzburg, 2020-07-07:
this will do..looks like a good start....some comments:
1. RX needs to bee generic block
2. Gimbal controller needs label as such with and (optional note)
3. Telem radio should be generic block (BT/SiK/WiFi/RX telem in)
4. Flip side and top side should have a note...maybe even gray the bottom side some....right now it leads one to think there are two autopilots...
5. Note that buzzer must be active not passive...maybe sneak it out a bit and make a little bigger with Active 5V Buzzer as its label?
6. Many ESCs can take more than 3S...I would drop that and just say Lipo Battery

I can add it to the wiki when you make the changes, just add it in another comment, or I can do them when time permits...

tamboekie, 2020-07-08:
Thank you for your comments. It seems that your goal is to depict a very generic setup. I can understand this, but I doubt that I have the knowledge needed to create a useful diagram like that. For example, by IBUS receiver is connected to RX2, but I think if I had a generic receiver, this could be connected to either the RX2 or SBUS pads, depending on the protocol. Another example may be the telemetry, which could technically be connected to any free UART, while the voltage supply may be quite different (perhaps another device needs VBAT or 3.3V and not 5V, and so on.

So, if I make the changes as you suggest (I agree to them all, BTW), will you (@Hwurzburg) be able to review it sufficiently, given that I do not have the other devices, nor experience with them? 

My real intent was to provide a single example to help others validate their own setup. I think my system would be typical for a low-cost build, but I am sure that the definition of low-cost will also be different depending on your frame of reference, so I now doubt whether the diagram is useful at all.

Some additional comments may be that I should remove the GND and 5V connections to reduce noise, do you agree? And perhaps other setups will have a BEC on the ESC, so again my discrete BEC would be optional, and should be marked as such.

Hwurzburg, 2020-07-08:
yes,  I will be able to review it for accuracy....btw, did you try the ibus on the sbus in pin?...it should work...AP is not like BF...it can decode everything except FPort on the SBUS input of flight controllers....the SBUS input is an inverter driving RX2 (but only if a signal is on SBUS)....we decode all inputs protocols , inverted or not, via an interupt that uses the RX2 in pin...in fact, there is an alternaate BRD_CONFIG that must be enabled to actually use that pin as a UART input rather than RC...

so any RX, including IBUS ones ,can directly attach to the SBUS pin

a wiring diagram with more than basic (ie no telem, no gimbal accessory) is fine... dont worry about the BEC, doesnt matter where is comes from, it still need one

tamboekie, 2020-07-26:
I've tried to update my diagram to be more generic and I'm sorry to say it just does not look professional enough for the Wiki. I think the problem is that each connected device needs to look a bit more detailed - I don't have the time to do this. Here's how far I got, but I proposed to bin the idea.

[image]

Hwurzburg, 2020-07-26:
[image]
Made a few tweaks....will add to wiki