ArduPilot/ardupilot_wiki #2909: Optical Flow: add information on how to use for non-gps navigation

rmackay9, 2020-07-08:
The optical flow pages specify how to connect an optical flow but they don't specify how to use it for non-GPS navigation.  The [Cheerson page](https://ardupilot.org/copter/docs/common-cheerson-cxof.html) has the most detail but these are the extra things people need to do:

- set ARMING_CHECKS so that the GPS check is disabled
- set EKx_GPS_TYPE = 3
- set the EKF origin using MP's Data screen, then right-mouse-button-click and select Set Home Here >> Set EKF Origin Here

Hwurzburg, 2020-07-08:
but there is an optical flow setup page that has all that, testing, first flights etc....it does NOT show the MP set home/origin though...

rmackay9, 2020-07-08:
@Hwurzburg, ah great, I didn't see the page at the end of the list is all, thanks!  Yes, adding the extra info about setting the EKF origin would be great.  To be clear, it's just the EKF origin that needs setting.  MP hides this "set EKF Origin Here" below the "Set Home" menu item but they're actually separate things.. and only the EKF origin needs setting, the setting of home will happen automatically immediately after the ekf origin is set.