ArduPilot/ardupilot_wiki #2911: Add instructions are setting up GSF

rmackay9, 2020-07-09:
ArduPilot 4.1 includes support for Gaussian Filter for estimating yaw, we should document this on the wiki

Hwurzburg, 2020-07-09:
what would the instructions be?

rmackay9, 2020-07-09:
I hope that @priseborough and/or @tridge can fill us in on the details but from what I know, I think the EKF3 might need to be enabled so that will require setting these params for now at least (when 4.1 goes out EKF3 may be the default):

- EK2_ENABLE = 0
- EK3_ENABLE = 1
- AHRS_EKF_TYPE = 3

I think for the emergency yaw reset feature it's always enabled and will takeover if the EKF becomes confused about it's yaw.

I suspect that the EK3_MAG_CAL param is used if there is no compass at all and we are only relying on GSF for yaw.

I've also created this video: https://www.youtube.com/watch?v=3xW9hj-lxNU

Hwurzburg, 2020-10-21:
think we can close this now

closed 2020-10-21 by Hwurzburg