ArduPilot/ardupilot_wiki #2922: Setting up the Build Environment (Linux/Ubuntu)

granthoss, 2020-07-18:
[Setting up the Build Environment (Linux/Ubuntu)](https://ardupilot.org/dev/docs/building-setup-linux.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Section: Cloning with the command line
git clone https://github.com/your-github-userid/ardupilot 

Does not work and leads to the error repository not found.

Should be fixed to the follow with proper capitalization of ArduPilot:
git clone https://github.com/your-github-userid/ArduPilot
cd ArduPilot

Hwurzburg, 2020-07-19:
the instructions are correct....you did not fork the Arudpilot/ardupilot repo as per the instructions to your repos on github....the command clones YOUR fork, but it has to exist first... https://ardupilot.org/dev/docs/git-fork.html

closed 2020-07-19 by Hwurzburg

granthoss, 2020-07-19:
That's right, not sure how I misread that on my side, thanks.