ArduPilot/ardupilot_wiki #2941: Auto Mode

Yulbrines, 2020-08-07:
[Auto Mode](https://ardupilot.org/copter/docs/auto-mode.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Yulbrines, 2020-08-07:
Sorry for my English, it's translated directly from google.
Recently I have started to practice with missions, after having managed to simulate a mission with a mission plan I have gone to the airfield but there I have not managed to get the auto mode to work, it simply ignores it, I connect the battery, I hope to have blocked satellites, I build engines with the alt hold mode and switch to auto but it does nothing, if I accelerate it takes off in alt hold mode, but auto does not connect, it is as if that flight mode did not exist. I would appreciate any suggestions please, thank you

rmackay9, 2020-08-07:
It's best to raise support question in the forums.  https://discuss.ardupilot.org/c/arducopter/copter-40/155

It's best to arm the vehicle in Loiter or Guided if you want to use Auto.

closed 2020-08-07 by rmackay9

Yulbrines, 2020-08-07:
Thanks for your quick response, I will take it into account in the next experience, but I also tried several alternatives in terms of flight mode (I can not specify because since it does not work, several tests are made without clear sense)