ArduPilot/ardupilot_wiki #2942: Remove Old Mavproxy docs

mnihalm, 2020-08-07:
## Bug report

**Issue details**

Mavproxy documentation is available in two places - 
 - https://ardupilot.org/mavproxy/index.html
 - https://ardupilot.github.io/MAVProxy/html/index.html

The former is up to date, but the latter is not. The outdated one has a higher SEO ranking and is the one most people would move to. 

I think the old one should be removed or should redirect to the new one.

Hwurzburg, 2020-08-07:
clicking either link goes to ardupilot.org....already redirected....

mnihalm, 2020-08-08:
Strange!
I could swear when I made the issue it wasn't the case. Found that it was outdated on gitter - https://gitter.im/ArduPilot/MAVProxy?at=5f2c83239231d665df5ed2b6

Looks like someone fixed it!, anyway that's great

mnihalm, 2020-08-08:
Looks like the redirect is not properly implemented? Google search results still have the old website higher in the rankings and now clicking on the results for anything other than the homepage gives a 404 

See - 
[image]

the 404 - 
[image]

I'm guessing this will get organically resolved as people start using the proper link more often

stephendade, 2020-08-10:
I put in a redirect from the old documentation to the ArduPilot wiki over the weekend. Appears that the redirect only works for root pages though (likehttps://ardupilot.github.io/MAVProxy/html/getting_started/index.html). Google (and other search engines) should update the urls in the near future.