ArduPilot/ardupilot_wiki #2986: HOTT Telemetry not working

Huege, 2020-08-26:
Dear all,

was trying to activate HOTT telemetry on ArduPlane and failed while following this setup guide: [HOTT Telemetry](https://ardupilot.org/copter/docs/common-hott-telemetry.html). Also tried out different ports. No sensors showing up at all.
My setup is F765 wing, GR12 + MZ 18 transmitter.

Thanks in advance for your support.

Hwurzburg, 2020-08-26:
please report this in the discuss.ardupilot.org forum.....if it turns out to be something missing or inaccurate in the wiki, we can change it...but the issue must be resolved first to assure that is the case...thanks

Huege, 2020-08-27:
done: https://discuss.ardupilot.org/t/graupner-hott-telemetry-doesnt-work-at-all/52995/22?u=guidohuelsmann

closed 2020-10-21 by Hwurzburg