ArduPilot/ardupilot_wiki #3016: Add links between FFT and "Batch Sampling" pages

rmackay9, 2020-09-10:
We've got these related pages which should be combined or at least we should add links between them:

- https://ardupilot.org/copter/docs/common-imu-fft.html
- https://ardupilot.org/copter/docs/common-imu-batchsampling.html

I think it's @peterbarker and maybe @andyp1per who are responsible for these two pages so perhaps they should pitch in their ideas on what to do.

Hwurzburg, 2020-09-10:
we also have https://ardupilot.org/copter/docs/common-imu-notch-filtering.html....
the batch sampling page speaks to generic noise analysis and is still a valid tool, although more direction towards using the following pages could be inserted now:
the imu-fft page is for upcoming in-flight FFT  feature which changes:
the imu-notch pages filter center

The titles of the pages could be simplified to be a bit more direct in some cases.... 
the notch filter doc already links to the batch sampling page...and the in-flight FFT obviates the need to use that, as I understand it....

so, I think:
- shorten the title on the inflight FFT page might be in order to avoid confusion with the basic notches page
- add directions to the notch and in-flight fft pages from the batch-sampling pages, now that we have those capabilities
- add the batch sampling page link as an additional noise analysis tool on the TUNING landing page

andyp1per, 2020-09-10:
Sounds good. A couple of points:
- not all boards will have in-flight FFT and not all users will want to use it since there is some CPU cost
- the biggest gripe I see from users is not understanding how to read the batch sampling page, maybe putting the relevant bits inline in the notch filtering page would help

Hwurzburg, 2021-12-22:
think this is resolved now with rer-org

closed 2021-12-22 by Hwurzburg