ArduPilot/ardupilot_wiki #3022: Getting and Setting Parameters

wucke13, 2020-09-11:
[Getting and Setting Parameters](https://ardupilot.org/dev/docs/mavlink-get-set-params.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

I just started writing a parser for the `apm.pdef.xml` files. Upon doing so, I stumbled across multiple bugs in https://autotest.ardupilot.org/Parameters/ArduCopter/apm.pdef.xml. Most issues are small oddities regarding the data:

+ there is  `<param ... user="Advanceds"></param>` in it (`LND_ACC_P_NSE`)
+ quite a few of the numbers have spaces around them (e.g. `<value code=" 1">KalmanFilter</value>`)

Is there a better source for getting information about the parameters?

khancyr, 2020-09-12:
Hello,
I am not sure that apm.pdef.xml is still use somewhere .... I will put an issue to either delete it or update it. 
Prefer the apm.pdef.json version. If json is not possible, you can use the xml that Mission Planner is using `--format xml_mp` instead of xml. You can also generate parameter for all vehicle with `https://github.com/ArduPilot/ardupilot/blob/master/Tools/scripts/generate_mp_paramfile.sh`

wucke13, 2020-09-13:
Oh no  :unamused:  I spent literally 8 hours on build a parser which somehow reads it. This information has to be presented way more prominently! If one searches for "ardupilot parameters" one of the first couple of links will be [this](https://ardupilot.org/dev/docs/mavlink-get-set-params.html) and in that document the XML files are recommended for GCS usage.

Where can I get `apm.pdef.json` ? Mission planner is not really an option for me, I'm on linux.

# TODO

+ Provide download of `apm.pdef.json` in a prominent manner
+ Mention the deprecation of  `apm.pdef.xm`

peterbarker, 2020-09-13:
I've created https://github.com/ArduPilot/ardupilot/pull/15331 to validate and correct the `@User` field.

I've created https://github.com/ArduPilot/ardupilot/pull/15330 to canonicalise the @Values field before attempting to validate it and further parse it in the various emitters.

The xml is in no way deprecated.  We continue to maintain it.

https://ardupilot.org/dev/docs/gcs-resources.html provides a link to the directory which contains the XML - but we don't currently generate the JSON on the server.  Perhaps after we get some more feedback on people of the format being used in there?

khancyr, 2020-09-14:
ok ... I thought that Tridge mention that it wasn't maintained when I made the MissionPlanner version.
Anyways, it can be fixed, and we could also apply the same formatter that I used for the missionplanner version, that way, it will be parsable easily.

wucke13, 2020-09-14:
> The xml is in no way deprecated. We continue to maintain it.

There is many bad things in that XMLs, making it super tedious to write a correct parser for it. I'm glad that this issues are addressed this quick, however!

khancyr, 2020-09-15:
I have made a PR to fix most whitespace issue and formatting : https://github.com/ArduPilot/ardupilot/pull/15347