ArduPilot/ardupilot_wiki #3067: Onboard Message Log Messages

AndKe, 2020-10-06:
[Onboard Message Log Messages](https://ardupilot.org/copter/docs/logmessages.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Please describe the difference between POS,AHR2   - when lat,lon from those is plotted on a map over each other, it may deviate a little, so there is some kind of difference there.

IamPete1, 2020-10-12:
POS comes from the primary EKF, and AHR2 comes from the DCM.

Hwurzburg, 2020-10-12:
wish we had enough bandwidth to fully explain these things, but just too many of them...like (my personal favorite) how do PIDx.tar and the various PIDx.P/I/D relate and interpreted....etc.

closed 2020-10-12 by Hwurzburg

peterbarker, 2020-10-12:
On Mon, 12 Oct 2020, Peter Hall wrote:

> POS comes from the primary EKF, and AHR2 comes from the DCM.

Not quite that simple, sadly.

AHR2 can get filled in by a second EKF...

We want a separate DCM message!

rmackay9, 2020-10-13:
@peterbarker,

if we are listing up our AHRS/EKF logging changes, I'd like to see the logging changed so we use an instance instead of having NKF1, NKF2, etc where NKF5 is arbitrary actually the same as NKF1 but for the 2nd EKF core.

peterbarker, 2020-10-13:
> @peterbarker,
> 
> if we are listing up our favourite AHRS/EKF logging changes, I'd like to see the logging changed so we use an instance instead of having NKF1, NKF2, etc where NKF5 is arbitrary actually the same as NKF1 but for the 2nd EKF core.

Unless I'm misunderstanding,  I did that about a year ago :-)

https://github.com/ArduPilot/ardupilot/pull/12997

In fact, NKF5 (and XKF5) are note the same as NKF1 (and XKF1) - it's a magic optical-flow message.  XKF6 *was* the same as XKF1 (it's not XKF1 with a core field value of 1.