ArduPilot/ardupilot_wiki #3111: Complete Parameter List

AndKe, 2020-10-30:
[Complete Parameter List](https://ardupilot.org/plane/docs/parameters.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

https://ardupilot.org/plane/docs/common-prearm-safety-checks.html#parameter-checks says "Parameters check - (unused)"
This is contradicted by https://ardupilot.org/plane/docs/common-prearm-safety-checks.html#parameter-checks

Hwurzburg, 2020-10-30:
you give two identical links

AndKe, 2020-10-30:
Thank you, I fixed it now.
Short version: parameters lists (for any platform) says the "parameters check" is unused, the other page, and source, contradicts it.

Hwurzburg, 2020-10-31:
that is not a wiki issue....there is an error in the codes metadata for that param.....when they actually started using it in Plane (for Quadplane params for ATT and POS control), they forgot to removed the unused tag...will open an issue in code side...then close this...thanks

Hwurzburg, 2020-10-31:
since you entered a code PR, closing this one

closed 2020-10-31 by Hwurzburg