ArduPilot/ardupilot_wiki #3117: Intel RealSense T265  Position drift in downward facing, and 45 degree tilt facing

SaiHaneeshAllu, 2020-11-01:
Following the setup in [Intel RealSense T265](https://ardupilot.org/copter/docs/common-vio-tracking-camera.html), we we're able to **fly the quadcopter well in loiter mode with T265 facing forward.** . But when the same attempt is made with T265 **facing Down , and 45 degree tilt** (orientation parameter changed accordingly in script) , **It works fine when quadcopter is disarmed** and is moved manually hand held. But as **soon as the quad is armed and slight throttle is given** to take it off, the **position starts drifting** and the messages are displayed as follows.

Position jumped by:  0.40215476176730836                                                                                                                                                                          INFO: VISO jump detected                                                                                                                                                                                          Position jumped by:  0.11720116021468974                                                                                                                                                                          INFO: VISO jump detected                                                                                                                                                                                          Position jumped by:  0.1959337960703581......
......

ArduCopter version -->  V4.0.3 (ffd08628)
Camera mounting is shown in below pictures

[image]
[image]

Any suggestions on how to resolve or possible reasons for the issue?

rmackay9, 2020-11-01:
@SaiHaneeshAllu,

thanks for the report.  The T265 is sensitive to vibrations so especially when facing downwards where it likely doesn't have as many features as it does when facing forward it loses it's position estimate.  Improved vibration dampening may help.  I'm afraid there probably is not much that we can do on the ArduPilot side.  Please note though that there are some robustness improvements in master (aka "latest") including allowing the use of barometer for the altitude estimate instead of always using the camera.

closed 2020-11-01 by rmackay9

SaiHaneeshAllu, 2020-11-02:
@rmackay9  Thanks for the clarification.