ArduPilot/ardupilot_wiki #3133: `TUNE_PARAM` documentation inconsistent for values above 100

arikrupnik, 2020-11-08:
### PART I

https://ardupilot.org/plane/docs/common-transmitter-tuning.html#tuning-multiple-parameters says:

> if you set TUNE_PARAM to 101 then you will have 4 different stages to your tune:
> 
> RATE_RLL_D
> RATE_RLL_PI
> RATE_PITCH_D
> RATE_PITCH_PI

A table lower down the page gives "params tuned" for `TUNE_PARAM=101` as
> Q_A_RAT_RLL_D/Q_A_RAT_RLL_PI & Q_A_RAT_RLL_I/ Q_A_RAT_PIT_D/Q_A_RAT_PIT_P & Q_A_RAT_RLL_I

It is unclear if there are four or six stages in the table definition--I'm unclear on how to interpret slashes vs. ampersands in that expression. Perhaps ampersand binds more tightly in this expression, meaning there are four stages:
* Q_A_RAT_RLL_D
* Q_A_RAT_RLL_PI & Q_A_RAT_RLL_I
* Q_A_RAT_PIT_D
* Q_A_RAT_PIT_P & Q_A_RAT_RLL_I

This agrees with the number of stages, but leaves me wondering what it means to tune `PI` and `I` in the same stage. Higher up on the same page it says, "notice that `RATE_RLL_PI` is actually two parameters in one, controlling both the P gain and the I gain for roll." I understand this to mean that tuning `RATE_RLL_PI` sets both `Q_A_RAT_RLL_P` and `Q_A_RAT_RLL_I` parameters. In this case it is unclear what additional information `& Q_A_RAT_RLL_I` adds in the table.

### PART II

The same table gives conjunctions of parameters for values 102 (roll), 103 (pitch), 104 (yaw) that have different sequences for each axis:

`rll: D, PI,I`
`pit: D, P, I`
`yaw: P, I, D`

If the different sets tune different parameters (or in different order), it would be helpful to explain the reasoning. An uninformed reader like me assumes that at least roll and pitch tuning follows the same sequence.

Hwurzburg, 2020-11-08:
agree it could be clearer....also I am surprised that the ref to Q_A_RAT_RLL_PI param did NOT generate a build error since the parameter is not named that but it hot links anyway...should be Q_A_RAT_RLL_P ...I will fix

pitch and roll QPIDs should have P and I equal....not true for Yaw....this is like Copter...explaining the reasoning is a bit beyond the scope of a setup guide but could be explained in the dev section on the PID loops (which is woefully lacking anyway and has many errors as the code changed over time)

arikrupnik, 2020-11-08:
> ref to Q_A_RAT_RLL_PI param did NOT generate a build error

The link is to `Q_A_RAT_RLL_P`, so looks like a copy-paste error. I can make a PR for this.

> pitch and roll QPIDs should have P and I equal....not true for Yaw

Assuming the `PI` in roll is a typo and should be `P`, there appear to be two differences:
* roll and pitch rate are two-step processes; yaw is a three-step, `P` and `I` are independent values
* yaw tunes the gains in a different order: `P`, `I`, `D` vs. `D`, `PI` for roll and pitch.

For the first bit, I think it's enough to say what you said, that it's `P` and `I` are the same for pitch and roll but separate for yaw, but the full explanation is outside the scope of this document.

For the other bit, the different order, I wonder if that's actually want happens, or if it is also a typographical error.

Hwurzburg, 2020-11-09:
#3135 addresses

closed 2020-11-09 by Hwurzburg

arikrupnik, 2020-11-09:
Can you confirm that `Set_RateYaw` tunes `P` first, unlike all the others that tune `D` first?

Hwurzburg, 2020-11-09:
https://github.com/ArduPilot/ardupilot/blob/68b6ecc59cd4f3ef93ce1ab47b69722f09956f97/ArduPlane/tuning.cpp#L29