ArduPilot/ardupilot_wiki #3134: Understanding SITL with RealFlight

KofiOwusu, 2020-11-08:
[Understanding SITL with RealFlight](https://ardupilot.org/dev/docs/flightaxis.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Dear Sir,

Recently got Realflight 9, inspire by Andrew Tridgell. Am currently learning SITL with RealFlight but having some issues. Am using two screens one running mission planner and the other running Realflight 9. Once i move the mouse over to the mission planner to upload my planned mission then Realflight 9 stops working , secondly any tips on resizing the screen just in case i want to use my laptop only.

Regards 

Kofi Owusu-Adusei

Hwurzburg, 2020-11-09:
this is a support issue, not a documentation error ...try discuss.ardupilot.org....thanks

that said read and follow : https://ardupilot.org/dev/docs/sitl-with-realflight.html

closed 2020-11-09 by Hwurzburg

KofiOwusu, 2020-12-27:
Thanks so much. I got it working