ArduPilot/ardupilot_wiki #3213: Outdated RTK info

Hwurzburg, 2020-12-16:
Randy, trying to update the wiki on RTK GPS...need some answers, with my intended actions:

1. apparently,, the Trimble is no longer made...archive it?(yes)
2. Septentrio is out of date...AsteRx-m is no longer made... remove from page?(yes)
3. JDrones has no GPS products ...remove them from landing page?(yes)
4. Holybro RTK is not mentioned on GPS for YAW page, shall I add it?(yes)
5. Should I add mro RTL link on landing page? (yes)

let me know

rmackay9, 2020-12-18:
@Hwurzburg

1. No, I think we should instead try and update with a more recent link.  I've ping'd the dev-team channel.
2. No, I think we should leave this one because the company is still around and maybe the same instructions work for newer models.  I'm not a fan of completely removing the page because it means users will assume this company's GPSs do not work with ArduPilot when they probably do.  I'm sure we have users (even core developers) using these GPSs.
3. yes
4. yes please!
5. yes please!

Thanks very much for this

khancyr, 2020-12-18:
about 2. AsteRx-m2 is not discontinued only the AsteRx-m, so we may remove the mention from AsteRx-m and keep the page for AsteRx-m2

rmackay9, 2020-12-18:
Great, so for 2, let's just remove the "-m" links from the page.

Hwurzburg, 2020-12-18:
Yeah that's what I meant in number two

Hwurzburg, 2020-12-31:
done...closing

closed 2020-12-31 by Hwurzburg