ArduPilot/ardupilot_wiki #3215: Change gitter links to discord

acxz, 2020-12-16:
It seems that most of the community has moved over to the discord server (https://ardupilot.org/discord).
The gitter channels are seeing less and less activity however several new users are still unfortunately finding their way there.

I believe it would be best to replace the gitter links, as shown [here](https://github.com/ArduPilot/ardupilot_wiki/search?q=gitter) in the documentation to the discord links.

Hwurzburg, 2020-12-17:
thanks...put on my todo list

Divyateja04, 2021-11-16:
Note: I sent a PR for this but the drone kit's gitter links still work so don't think we need to remove those

Hwurzburg, 2021-11-16:
closing....the links to qgc stuff and mavlink should remain as they are not ArduPilot administered and should be used...the last two I have fixed...

closed 2021-11-16 by Hwurzburg

Divyateja04, 2021-11-16:
Yeah but those links don't work, even the qgc one shouldn't we update them...and also I saw some gitter links for ardupilot in the dev docs, I updated those