ArduPilot/ardupilot_wiki #3219: Communication Channels

AndreHenke, 2020-12-18:
Tricopter motors configuration is wrong.
You should swap motors 2 and 4 in order to fly, or you will crash.

rmackay9, 2020-12-18:
This is the wiki to-do list so it's probably best to report this in the forums first and then if it's a real issue we can raise it in the flight code issues list. https://github.com/ArduPilot/ardupilot/issues

Sorry to be annoying but just trying to keep our to-do list separated from support issues.

closed 2020-12-18 by rmackay9

rmackay9, 2020-12-18:
@AndreHenke,

I've checked with some other members of the dev team and we think the motor configuration document on the wiki is correct.  If you've got a log or something and can post it in the support forums and then drop a link here we can have a peek. txs!