ArduPilot/ardupilot_wiki #3249: hglrc zeus f722

Sderby1229, 2021-01-04:
[Search](https://ardupilot.org/dev/search.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

rmackay9, 2021-01-04:
The description isn't very clear but I assume this is a request to add support for thhe hglrc zeus f722 flight controller.  https://www.banggood.com/20mm-HGLRC-Zeus-F722-mini-MPU6000-3-6S-F7-betaflight-Flight-Controller-w-or-OSD-Barometer-BLACKBOX-5UARTS-For-DJI-Air-Unit-Caddx-vista-FPV-Racing-RC-Drone-Freestyle-Quad-p-1651471.html?akmClientCountry=America&cur_warehouse=CN&utm_source=google&utm_medium=cpc_ods&utm_campaign=nolan-sds-broad-motor-all-1222&utm_content=nolan&ad_id=487900326239&gclid=Cj0KCQiAlsv_BRDtARIsAHMGVSaKqWAlvb8gJ8OThsLymBy5dCDB7QFDZdZb6WcLJJF_yw2VnLjlIioaAp22EALw_wcB

Hwurzburg, 2021-01-05:
sorry but this controller does not have enough flash space (512K), however, this 20x20 controller IS supported:
https://www.racedayquads.com/products/racerstar-f405-wing-nano-20x20-flight-controller-w-ardupilot-inav-super-bluetooth-qgroundcontrol?variant=32253089382513&gclid=CjwKCAiAudD_BRBXEiwAudakXwbNxZiHgoVoIGlmHPQmeO9umAFwGu1zxz4zBkYk1Y58V3nPuOmYrRoCHB8QAvD_BwE

 and in the future you can make feature and enhancement requests in the code section of Github for ArduPilot by opening an issue there (https://github.com/ArduPilot/ardupilot/issues) instead of here  in the wiki...

thanks....closing this

closed 2021-01-05 by Hwurzburg