ArduPilot/ardupilot_wiki #3261: Boat Configuration

biomestry, 2021-01-07:
[Boat Configuration](https://ardupilot.org/rover/docs/boat-configuration.html) - when i flash ardurover on to my pixhawk 4 there's no com ports i can connect to on mission planner. In my device manager it only shows 1 port pixhawk4 and none on mission planner for me to connect to. This issue only happens with ardurover, I've tried flashing arducopter and plane and i was able to connect and setup.
[image]
[image]

rmackay9, 2021-01-07:
I wonder if perhaps this is because the Rover firmware is a little older than the plane and copter.  Could you try loading "latest" (aka master) onto the board by pressing Ctrl-Q while on the Install Firmware screen?  It should change the label below the rover icon to be "Rover 4.1.0 DEV".  If this works then we know that the issue will be resolved with Rover-4.1 which should start beta testing within a month.

biomestry, 2021-01-07:
@rmackay9  ohh okayyy, i'll give it a shot, but i've seen people make a boat with a pixhawk and ardurover so idk why i'm having this issue. but thanks

biomestry, 2021-01-07:
@rmackay9 hey heyyy, thank you very much for your 2 cents, i was able to connect after i used rover4.1dev it worked like a charm and then i changed some parameters to check if the motors are working and they are working. literally been trying to fix this issue for the month.

rmackay9, 2021-01-08:
@biomestry excellent.  Hopefully we can get 4.1 out into beta testing in the near-ish future.

closed 2021-01-08 by rmackay9