ArduPilot/ardupilot_wiki #3273: combine or group FFT and Harmonic Notch pages

rmackay9, 2021-01-13:
I think we should combine or group these two pages under a landing page:

- [Managing Gyro Noise with the Static Notch and Dynamic Harmonic Notch Filters](https://ardupilot.org/copter/docs/common-imu-notch-filtering.html)
- [Managing Gyro Noise with In-Flight FFT](https://ardupilot.org/copter/docs/common-imu-fft.html)

I'm definitely not an expert in this harmonic notch and FFT feature but I think these two pages have duplicate information and should be rationalised somethow.

FYI @andyp1per

hendjoshsr71, 2021-11-27:
Just a poke. We have had a user mention issues about these pages and difficulty in parsing them as a whole idea as stated in this issue.

Hwurzburg, 2021-12-22:
done

closed 2021-12-22 by Hwurzburg