ArduPilot/ardupilot_wiki #3292: Ready-To-Use --  Makeflyeasy

patrickpoirier51, 2021-01-23:
Looking at Aliexpress , most of VTOL Planes are using Ardupilot

Example:
Makeflyeasy Fighter (VTOL Version)  
https://www.aliexpress.com/item/4001113285374.html
https://www.aliexpress.com/item/1005001858282219.html

Including a ''always free'' Ground Station...

[image]

Hwurzburg, 2021-04-08:
actually I see nothing that indicates these are ArduPilot based rather than PX4 or some custom firmware...only that they seem to have MAVLink telem....until this is confirmed we should not list them...

rmackay9, 2021-04-08:
If it's mission planner then it's almost guaranteed to be AP (although I have bumped into one example where a custom firmware used MP).

patrickpoirier51, 2021-10-13:
Thanks for updating @makeflyeasy  :-)

Hwurzburg, 2021-11-13:
MakeFlyEasy vehicles added to wiki...closing

closed 2021-11-13 by Hwurzburg