ArduPilot/ardupilot_wiki #3293: BLHeli_32 Pass-Through Support

Turf1975, 2021-01-23:
[BLHeli_32 Pass-Through Support](https://ardupilot.org/copter/docs/common-blheli32-passthru.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.
Dear developer team,
In my case I had to push the safety switch before the pass-through was working. Maybe this could added as a remark in the documentation.
I am running arducopter 4.05 with hakrc blheli32 escs.
Best regards,

Maarten

Hwurzburg, 2021-01-23:
Randy, please confirm and I will add...thanks

Hwurzburg, 2021-02-04:
@rmackay9 bump

rmackay9, 2021-02-05:
@Hwurzburg, I'm afraid that I really don't know because I don't use BLHeli ESCs very often.  I think I've used them only once and that was on a boat.  Probably @tridge or @andyp1per is in a better place to respond.

andyp1per, 2021-02-05:
Yes, pretty sure this is true. I have a PR in the works that updates the passthru docs for BLHeli_S as well

Hwurzburg, 2021-04-08:
@andyp1per did you add this? I dont see it in any of the docs...if not, please confirm that we should add a note to the BL_Heli passthrough page

davidbitton, 2021-04-11:
@andyp1per it's not working on my Matek H743-SLIM running master.

andyp1per, 2021-04-11:
What ESC? It’s working for me on this board

davidbitton, 2021-04-11:
T-Motor F55A Pro II F3 32bit 55A 3-6S 30x30 4in1 ESC w/ BEC

Hwurzburg, 2021-11-13:
@andyp1per should I add a note in the warning section that safety switch must be off, if used, in order to communicate via passthru?

andyp1per, 2021-11-13:
@Hwurzburg good idea!

andyp1per, 2021-11-13:
I actually fixed this on the F55 so closing

davidbitton, 2021-11-13:
> @andyp1per should I add a note in the warning section that safety switch must be off, if used, in order to communicate via passthru?

Does that apply to situations where there is no safety switch; e.g. when `BRD_SAFETYENABLE` is set to `0`?

Hwurzburg, 2021-11-13:
no