ArduPilot/ardupilot_wiki #3304: Mission Commands

emin1461, 2021-01-26:
[Mission Commands](https://ardupilot.org/planner/docs/common-mavlink-mission-command-messages-mav_cmd.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

MAV_CMD_DO_SET_HOME this command is not showing on mission planner .

rmackay9, 2021-01-26:
This is actually a mission planner issue so it should really be raised in the Mission Planner issues list.  https://github.com/ArduPilot/MissionPlanner/issues

Can I ask why you'd want to set the home position as part of a mission?  We support it but I've always wondered what the use case could be.

closed 2021-01-26 by rmackay9

emin1461, 2021-01-26:
> This is actually a mission planner issue so it should really be raised in the Mission Planner issues list. https://github.com/ArduPilot/MissionPlanner/issues
> 
> Can I ask why you'd want to set the home position as part of a mission? We support it but I've always wondered what the use case could be.

I'm just learning. I translate the meaning of the commands written there into my language and examine them in the mission planner. I wanted to write when I couldn't see this command on mission planner. I don't understand exactly what happened  :)