ArduPilot/ardupilot_wiki #3311: Download and Installation -  missing library

AndKe, 2021-02-01:
[Download and Installation](https://ardupilot.org/mavproxy/docs/getting_started/download_and_installation.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

please add libsfml-dev to the apt install lists, this library is required for --osd to work

peterbarker, 2021-02-01:
We don't install every Python module which is required to run all of the MAVProxy modules.  People that are installing MAVProxy on a companion computer for packet routing purposes wouldn't thank us for doing so.

It is installed as part of the ArduPilot developer environment install scripts - and that's where the  `--osd` options comes into it, I believe.

Is there some reason the env install scripts aren't sufficient here?

AndKe, 2021-02-02:
A colleague discovered that it did not work for him in the field.
By my observation, some MAVProxy users use it for both packet routing and GCS (simultaneously), hence not all have ardupilot environment set up. 
So it is kind of logical that at least the "lighter" modules should work "out of the box" - (not necessarily the cesium plugin)
My request is just to make the experience more streamlined.

khancyr, 2021-02-02:
Mavproxy don't use libsfml and don't have --osd option . That is a SITL option so the related packages are already in the install scripts.

Unless I missread something

AndKe, 2021-02-02:
@khancyr  You are right, I've never given that a thought :)   to me, --osd is something I use with SITL.   My coworker picked up --osd from my example :) 
let me close this issue in shame :)

closed 2021-02-02 by AndKe

khancyr, 2021-02-02:
no worries, there is an osd module on mavproxy but that only for playing with OSD parameter.