ArduPilot/ardupilot_wiki #3320: DShot: how to set trim to 1500 instead of 1000

wucke13, 2021-02-07:
The [documentation for DShot](https://ardupilot.org/copter/docs/common-dshot.html#dshot) mentions the following:

> When an output is configured for DShot, the SERVOx_MIN/MAX parameters for that output will always be reset to 1000/2000, respectively, upon every boot, since DShot does not use these parameters and they can affect scaling in other sections of the code. The SERVOx_TRIM value will be reset to 1500 if it’s a reversible output, or 1000 if normal output setup in DShot.

However, I could not find any information on __how or where__ one can determine whether the output is reversible, or not.

Hwurzburg, 2021-02-07:
simply use the SERACH box in the upper left corner for reverse thrust....

wucke13, 2021-02-07:
@Hwurzburg I did look through [Reverse Thrust Setup](https://ardupilot.org/plane/docs/reverse-thrust-setup.html), if you mean that. Would you mind telling me which parameter one has to change? It is not `THR_MIN`, I've set it to -100 and still get 1000 as TRIM value for the servo channel in question.

Hwurzburg, 2021-02-07:
sorry if it came across that way.....I was not aware that you could not adjust the TRIM if DSHOT was enabled and believed they behaved as would any other reversible ESC,...I had just added that  REVERSE THRUST section a few months ago and am surprised that the review process did not catch that DSHOT escs would prevent changing the TRIM value in addition to the endpoints....will discuss with the coder to see what impact REMASK has besides allowing the setting of the TRIM value (I infer this although you do not state directly this occurs) and try to expand the wiki as appropriate...

wucke13, 2021-02-07:
@Hwurzburg Hi, thank you for your honest answer :heart:. AFAIK `SERVO_BLH_REMASK` does not allow the `SERVOx_TRIM` to be changed (in DShot, it's never possible to change it, I guess). However, without `SERVO_BLH_REMASK` the `SERVOx_TRIM` is pinpointed to 1000, but with a 1 set in `SERVO_BLH_REMASK` the `SERVOx_TRIM` in question will be pinpointed to 1500.

So, for reverse thrust to work on any DShot channel, the bit has to be set. Otherwise the ESC will do some uint to int cast magic resulting in a weird throttle curve, which with ascending throttle value will first rise to maximum forward speed, and then jump from there to minimum backward speed with increasing speed if throttle goes up further. Consider this screenshot, I hope it makes it clear.

[image]

The x-axis is throttle input (`RC3`), the y-axis is observed motor speed (positiv for forward, negative for backward).

Hwurzburg, 2021-02-08:
I found the curve conversion in the code based on that mask param...it appears that it does not update the servo output params, just ignores them,  and sends 47 to 2047 to the esc...but AP may be doing something to them in a another motors module...trying to get some time with the coder to discuss... 
so throttle low is full rev, decreasing to 0 at midstick and then increasing to full forward at high stick after conversion..just like other reversing escs...THR_MIN/MAX and RC3_TRIM work as described in the wiki, but the user has no control over SERVO_MIN/MAX/TRIM when Dshot is being used and needs to evoke the conversion scaling with REMASK....

need to add to the existing note re SERVOx_TRIM being ignored also...and a note that you have to use the REMASK to be able to use reversing DShot escs.... may expand the param metadata in code itself also for _REMASK to explain what it does ,ie converting AP output to be compatible with the esc I/O curve in that mode...finally add a note in the DSHOT page also, might avoid a surprise to someone who just programs the esc for reversing and not the REMASK...

wucke13, 2021-02-08:
> so throttle low is full rev, decreasing to 0 at midstick and then increasing to full forward at high stick after conversion.

Does not match my observations at all, very interesting! Just a few notes:

The parameter, which defines whether Reverse Thrust is active in general is (so the answer to my first question)

[`THR_MIN` < 0](https://github.com/ArduPilot/ardupilot/blob/29775f310e62f5a22d0af23a2d6541ad0b899c38/ArduPlane/reverse_thrust.cpp#L109)

Hwurzburg, 2021-02-08:
THR_MIN<0....I think that is covered in the REVERSE THRUST article pretty well:
"... THR_MIN must be set to a negative value and will constrain the maximum reverse thrust at low stick in modes other than MANUAL to a percentage of the reverse output range. "

not sure what "does not match your observations" means....if you set REMASK, THR_MIN=-100 and THR_MAX=100 with a reversible DShot ESC, low thorttle should be full reverse, center off, high throttle max forward....that does not match you experience? what does it do?

wucke13, 2021-02-08:
> THR_MIN<0....I think that is covered in the REVERSE THRUST article pretty well:

I do agree. This is purely context for my initial question (and not a critique for the docs), in case anyone comes to read this in the future. I just don't want to create the "Somebody else had the problem ...  solved it ...  did not state what the solution was" scenario.

> not sure what "does not match your observations" means

My comment was purely related to the situation when `SERVO_BLH_REMASK` is not set. If the `SERVO_BLH_REMASK` is set correctly, I get exactly the behavior you described :+1: : 

> if you set REMASK, THR_MIN=-100 and THR_MAX=100 with a reversible DShot ESC, low thorttle should be full reverse, center off, high throttle max forward

Hwurzburg, 2021-02-08:
great....but playing around with this I have found some anomalies that I need to explore with the coders...
-Q_M_PWM_MAX/MIN set to 0 does not reflect RC3_MIN/MAX and outputs are set to 1000/2000 even in PWM types
-the output limit/trim params are overridden in DShot but params displayed changed only if a BLH mask has been set by BLH_AUTO ...Dshot and BLH are supposedly independent things
-you can rudder arm in plane while not at idle (mid stick or low if reverse switch is used)....a warning is needed at least
-if you set up a copter motor in plane as reversible with REMASK, boot unarmed sets the output to 1000 which is full reverse...at least it appears that it does in the output display...bit problem I think....in plane on a normal forward motor when setup for reverse it initializes at 1500us

wucke13, 2021-02-08:
Good that you found all these, great potential for improvement :slightly_smiling_face: !

Hwurzburg, 2021-03-15:
closing...I updated the note re the params for MIN/MAX, since they are not updated unless the BLHeli mask has been set, even if they use 1000/2000.....its a minor bug that someone needs to fix in the code...

closed 2021-03-15 by Hwurzburg