ArduPilot/ardupilot_wiki #3321: Theme update

khancyr, 2021-02-09:
Just to have it tagged to next dev call : Here is a PR to update our wiki theme : 
https://github.com/ArduPilot/sphinx_rtd_theme/pull/15

Update to the latest release
Disable npm dependencies as they aren't needed.
Squash our change in a single commit for easier management
Split our css change into a new ardupilot.css file.
Remove swag shop links

What does it solve :
numerous fix in JS, css and performances

Release process :
backup current master branch, force push this branch on master. Then the wiki need to reinstall the theme with pip

khancyr, 2021-02-09:
To test it, modify the Sphinxsetup.sh to point on my theme fork : 
`pip install git+https://github.com/khancyr/sphinx_rtd_theme.git -UI ||  pip3 install git+https://github.com/khancyr/sphinx_rtd_theme.git -UI`

And rebuild the wiki as usual.

rmackay9, 2021-02-15:
@khancyr can you provide some screen shots of what it looks like?

Is there a way to build locally?

khancyr, 2021-02-15:
@rmackay9 sure

Pr

https://cdn.discordapp.com/attachments/710994174299603055/808640825126158356/Capture_decran_de_2021-02-09_11-09-36.png

Master

https://cdn.discordapp.com/attachments/710994174299603055/808640941576552478/Capture_decran_de_2021-02-09_11-10-08.png

On Mobile 
Pr 

https://cdn.discordapp.com/attachments/710994174299603055/808641641464201216/Capture_decran_de_2021-02-09_11-12-59.png

Master

https://cdn.discordapp.com/attachments/710994174299603055/808641708496912384/Capture_decran_de_2021-02-09_11-13-16.png

To try it, you need to install the theme from my repo with pip. Then just rebuild the wiki as usual.
If you are using vagrant or docker, update the Sphinxsetup to install the theme from my repo, and rebuild the wiki.

Hwurzburg, 2021-02-21:
tried the new theme locally and cant see any width difference...switched to my local wiki master branch, installed as directed, rebuilt plane wiki...screen width the same as before...thought it was going to default to wider format...

khancyr, 2021-02-24:
@Hwurzburg I didn't enable the width change for now as I prefer that we do the upgrade first.
That is present in ardupilot.css, you just need to uncoment the commented lines !