ArduPilot/ardupilot_wiki #3327: SITL Simulator (Software in the Loop) - Wrong serial/tcp port for mavproxy on illustration.

trygve55, 2021-02-12:
[SITL Simulator (Software in the Loop)](https://ardupilot.org/dev/docs/sitl-simulator-software-in-the-loop.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Mavproxy seams to connect to serial 1(port:5760) instead of serial 3(port:5763).

Running on Ubuntu 18.04 set up with the install script for the build environment.

https://ardupilot.org/dev/docs/sitl-simulator-software-in-the-loop.html

khancyr, 2021-02-12:
hello,

the picture isn't wrong. you can connect multiple mavproxy instance.

closed 2021-02-12 by khancyr

trygve55, 2021-02-16:
Hello, I checked the source code for SITL and found that mavproxy is started using port 5760.

Link to ArduPilot repo:
https://github.com/ArduPilot/ardupilot/blob/master/Tools/autotest/sim_vehicle.py#L767