ArduPilot/ardupilot_wiki #3331: missing bit of info on page for Pixhawk FMUv2 / Cube FMU v3 Serial Mapping

mattdholmes00, 2021-02-15:
there's a column titled "SERIAL_CONTROL*" that is missing the explanation indicated by the "*" and it's not obvious to me what that column means.  I would fix it myself if I knew...

Hwurzburg, 2021-02-21:
@peterbarker Peter, what do those numbers mean??

peterbarker, 2021-02-21:
@Hwurzburg bizarrely my local copy of that document has this note at the end of it:

```
.. note:

   SERIAL_CONTROL is used to talk from a GS directly to a serially-attached device.

   On MAVProxy one can use ``module load nsh`` followied by ``nsh port 0`` to talk to a Hayes modem connected to TELEM1
```

That's not appearing on the page, and does answer @mattdholmes00 's query.

My tree isn't dirty - could you check that file yourself and see if the note is present?

(`**` is still dangling, mind you :-) )

Hwurzburg, 2021-02-22:
the note is miss-formatted and Sphinx throws it out I think...will fix

Hwurzburg, 2021-02-24:
fixed..closing

closed 2021-02-24 by Hwurzburg