ArduPilot/ardupilot_wiki #3372: Duplicate Aux switch option

Henryychua, 2021-03-08:
[Search](https://ardupilot.org/copter/search.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Henryychua, 2021-03-08:
Pixhawk unable to arm with prearm check message : Duplicate Aux Switch option, what cause it, and how to resolve?

rmackay9, 2021-03-08:
@Henryychua,

I think if you check the RCx_OPTION parameter values you'll find that two switches have been set to have the same auxiliary function.  I.e. RC7_OPTION and RC8_OPTION have the same value.

This is really a support issue so I'll close this.

closed 2021-03-08 by rmackay9

Henryychua, 2021-03-08:
RC7 & RC8 disabled but still give the same response, only way to stop it was to stop prearm check on RC which means I cannot set to check all but have to set individual items

Henryychua, 2021-03-08:
Thank for your response, RC7 and RC8 were disabled and still give same response, had to disable prearm check for RC and not set for all but individually

From: Randy Mackay
Sent: Monday, March 8, 2021 5:12 PM
To: ArduPilot/ardupilot_wiki
Cc: Henryychua; Mention
Subject: Re: [ArduPilot/ardupilot_wiki] Duplicate Aux switch option (#3372)

@Henryychua,
I think if you check the RCx_OPTION parameter values you'll find that two switches have been set to have the same auxiliary function. I.e. RC7_OPTION and RC8_OPTION have the same value.
This is really a support issue so I'll close this.

rmackay9, 2021-03-08:
@Henryychua,

AP4.0 (and higher) support many auxiliary switches so it might not be RC7 and RC8, that was just an example.