ArduPilot/ardupilot_wiki #3375: Wrong gyro units in "Downloading and Analyzing Data Logs in Mission Planner"

Mahdi-Hosseinali, 2021-03-10:
[Downloading and Analyzing Data Logs in Mission Planner](https://ardupilot.org/copter/docs/common-downloading-and-analyzing-data-logs-in-mission-planner.html)

I am working with some logs from ArduPilot. The GyrX units are definitely not degree/s. When I calculate the gyro rates by taking the derivative of the tilt angles, the gyro rates seem to have a unit of rad/(Pi/2)/s which sounds weird.

I tried to find units in the mavlink message from the files but nothing is there.

It is stated in this [old discussion](https://discuss.ardupilot.org/t/imu-units-of-the-pixhalk-log/19458/5) that the units are rad/s.

I'm not sure what the units are, they just don't seem to be degree/s so the page can confuse people.

hendjoshsr71, 2021-03-10:
The rotation rates are logged in logfiles as radians/seconds in both the GYR and IMU message.

https://github.com/ArduPilot/ardupilot/blob/f78d7c3b250e7a78b185f42b8c483b7566cd82b6/libraries/AP_Logger/LogStructure.h#L1520

Hwurzburg, 2021-03-14:
updated wiki...closing...thanks

closed 2021-03-14 by Hwurzburg