ArduPilot/ardupilot_wiki #3388: Sonar based on the sensor " JSN-SRO4T"

AlksSAV, 2021-03-17:
Hello. Many people use this sensor in their boats to deliver bait for fish. The STM32-based autopilot is used. I would also like to use it with Pixhawk. This sensor has several operating modes, the principle is the same as that of the HC-SR04 sonar. What parameters do I need to connect to the SERIAL 4/5 port? 
[image]
Attached is a file with a description of the operating modes. There is a TTL mode (mod 3)
[JSN-SR04T-2.0.pdf](https://github.com/ArduPilot/ardupilot_wiki/files/6156964/JSN-SR04T-2.0.pdf)

Hwurzburg, 2021-03-17:
please aks support questions in discuss.ardupilot.org....this is for wiki errors or additions only

closed 2021-03-17 by Hwurzburg

rmackay9, 2021-03-17:
@AlksSAV,

As Henry says this should go into the [Rover discussion forums](https://discuss.ardupilot.org/c/ardurover/rover-4-0/154).

In any case, there HC-SR04 wiki page is here including connection instructions: https://ardupilot.org/rover/docs/common-rangefinder-hcsr04.html

Support for the HC-SR04 is included in Rover-4.1 which hasn't started beta testing yet so if you want to test this you'll need to load "latest" onto the autopilot.  This can be done using MP's Install Firmware page.  Press Ctrl-Q and hopefully you'll see the label under the car icons change to, "Rover 4.1.0 DEV".

If you choose to test with "latest" please remember that this software has not gone through beta testing so it very likely has some issues.