ArduPilot/ardupilot_wiki #3394: Proximity/Avoidance documentation left

rishabsingh3003, 2021-03-19:
Its been a good year for proximity and avoidance libraries but unfortunately I have not kept up with documentation. Its very outdated now and it should be ready for 4.1 release. 
I am making a list below to make sure we cover everything (it'll be awesome if someone helps me out) :

- [ ] ArduPilot/ardupilot/pull/14623: How to tune BendyRuler so that it doesn't switch decisions too quickly and improve performance 

- [ ] ArduPilot/ardupilot/pull/14726:  Vertical BendyRuler

- [ ] ArduPilot/ardupilot/pull/14934: BendyRuler + Dijkstras fusion

- [ ] ArduPilot/ardupilot/pull/15136: Reject OA DB obstacles near home

- [ ] ArduPilot/ardupilot/pull/14819 : Simple Avoidance - Backing away

- [ ] ArduPilot/ardupilot/pull/15553: 3D simple avoidance (Currently only works 2D with loiter), 3D proximity library

- [ ] ArduPilot/ardupilot/pull/16500: Min Alt before avoidance is switched on

- [ ] ArduPilot/ardupilot/pull/16500: Max accel for Simple avoidance

- [ ] ArduPilot/ardupilot/pull/16604: Low Pass Filter (merged) and how to change cutoff freq (yet to be merged to master)

- [ ] ArduPilot/ardupilot/pull/16697: Filter land with downward facing rangefinder 

- [ ] Some updates to the developer documentation https://ardupilot.org/dev/docs/code-overview-object-avoidance.html

Please write down in the comments if I have missed something here. I'll be starting on this list soon.

rishabsingh3003, 2021-04-26:
https://github.com/ArduPilot/ardupilot_wiki/pull/3491 fulfils a lot on the list.

timtuxworth, 2021-12-30:
This is very interesting. What would it take to implement this in Plane?