ArduPilot/ardupilot_wiki #3407: Measuring Vibration

AvtechScientific, 2021-03-29:
[Measuring Vibration](https://ardupilot.org/copter/docs/common-measuring-vibration.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Please, provide all the docs, also as an e-book in pdf format for offline reading.

Hwurzburg, 2021-04-10:
Bruno, thanks for looking into this!

brunoolivieri, 2021-04-14:
There was some discussion some time ago regarding providing an offline readable wiki. So far as I remember, there were some issues, but I cannot found the historic.

I am inserting it in the roadmap to check how Sphinx generates that.

peterbarker, 2021-04-29:
https://docs.typo3.org/typo3cms/extensions/sphinx/2.5.1/AdvancedUsersManual/RenderingPdf/Index.html

darigovresearch, 2021-05-22:
Would definitely agree that having an autogenerated download version of the docs is helpful particularly as a backup when working on in a remote location where there is little to no internet connectivity

TunaLobster, 2021-06-30:
I got up to the point of doing: 
```
cd <wiki directory>
make latexpdf
```
It hits an functools.partial __name__ attribute exception when trying to write a vimeo video to a pdf (imagine that). Log is attached. Running in WSL Ubuntu 18.04. Taking the html with us is working well, except the search box doesn't work. That's a different rabbit hole though.
[sphinx-err-pvcoi057.log](https://github.com/ArduPilot/ardupilot_wiki/files/6743537/sphinx-err-pvcoi057.log)

Hwurzburg, 2021-11-13:
@TunaLobster update?

TunaLobster, 2021-11-13:
Not going to be able to do this until the environment is stable and can be easily updated on the build server. Peter has a branch for the new build server that I've made a PR too (the python venv and such). I know he's pretty busy so I'm just waiting on a review from him on that stuff. I know Pierre will want to look at the PR that brings Peter's branch upstream.

Hwurzburg, 2026-10-17:
I think this has been effectively addressed with the new downloadable wiki,,,closing

closed 2026-10-17 by Hwurzburg