ArduPilot/ardupilot_wiki #3425: Non-vectored Single Motor Tailsitter Planes: Any example?

songkuan79, 2021-04-07:
Hi,  I read in the arduplane wiki section that it can support Non-vectored Single Motor Tailsitter Planes.  This is quite new to me.  The video examples in the wiki section shows twin-motor tailsitters.  So I can't help wondering what would a single motor tailsitter look like and how many control surfaces  (2/3/4?) it takes.  I would really appreciate someone to share some experience on that.

I have an AR Wing, which has one motor and sits vertically on its winglet tips.  I just love to imagine it soars up sky vertically, maybe with twists just to stabilize itself vertically?

[Tailsitter Planes](https://ardupilot.org/plane/docs/guide-tailsitter.html) - Describe the issue/suggestion and improve the title. Please keep a link to the original article if relevant.

Hwurzburg, 2021-04-08:
it refers primarily to any normal 3d style plane with control surfaces large enough to allow stable hovering.....setup as a tailsitter with ArduPilot, hands off hovers and loitering in place are possible.....vertical takeoffs and landing (assuming a tail stand setup) are also possible, but with no wind, otherwise it will fall over when on the ground or landing...very stable tail stands can takeoff with some wind, but landing is a big issue....it is much easier with a twin motor vectored tilt setup....or as a TVBS (same thing but take off and landing from horizontal staring position)

songkuan79, 2021-04-09:
much appreciated! a tutorial very easy to understand.

Hwurzburg, 2021-04-09:
added some more notes on tailsitter page

closed 2021-04-09 by Hwurzburg