ArduPilot/ardupilot_wiki #3430: Rover: vectored thrust page update for 4.1

rmackay9, 2021-04-08:
PR https://github.com/ArduPilot/ardupilot/pull/17060 replaced the old parameter (MOT_VEC_SCALE?) with MOT_VEC_ANGLEMAX so we should update the vectored thrust page (https://ardupilot.org/rover/docs/rover-vectored-thrust.html) to tell users to update this new parameter instead.  The parameter is the angle (in degrees) of the motor's maximum deflection from center capped at 90deg.

Hwurzburg, 2021-04-08:
I have fixed the build error, but the entire page needs rework since the recommendation and graphs will have changed....have added it to the 4.1 list with you as author

rmackay9, 2021-04-12:
I've updated this page with instructions for 4.1.  I've left the 4.0 instructions in place but it's still the stable version.  I know @Hwurzburg doesn't like to have the version numbers in the page so I'm open to other ideas.

closed 2021-04-12 by rmackay9