ArduPilot/ardupilot_wiki #3434: MSP OSD: Distance Traveled not found

Target0815, 2021-04-08:
Hi Henry,

the entry "Distance Traveled" is listed in the wiki for the MSP OSD, but it is not to be found in the OSD configuration and I also think that it does not exist.

Hwurzburg, 2021-04-08:
I have asked the author to comment...dont know if this is really a wiki issue or support/bug, but will leave it here for now

Target0815, 2021-04-08:
Yes, @vierfuffzig certainly knows that, could have asked him himself ;)

vierfuffzig, 2021-04-09:
@Hwurzburg @Target0815 i'm sorry to say i'm neither the author of the code, neither of the wiki entry, both would be alex @yaapu  actually.
however i've checked the code and tested on SITL. imho this is a wiki issue, as (per code) "distance travelled" is not among the supported MSP OSD items. 

here's the list of supported MSP OSD items: https://github.com/ArduPilot/ardupilot/blob/c83774a7cd83929b90c772beec268d0fc1bf94d2/libraries/AP_OSD/AP_OSD_MSP.cpp#L22

i can confirm these working as expected on SITL:
[image]

additionally, blhtemp and fltmode (=failsafe-status) supposedly are supported too. i can't actually confirm that as right now i don't have actual hardware to test blheli + MSP OSD on. the fltmode panel does not show up on SITL MSP OSD emulation.

maybe alex can confrim - or add the support for the total distance item ;)

cheer, basti.

Hwurzburg, 2021-04-09:
duh....sorry I miss-remembered @yaapu ....do you want to PR a change to the MSP OSD page or shall I?

yaapu, 2021-04-13:
total distance is not supported by the DJI FPV Goggles V1, not sure if V2 are any different, don't have access to a set.
BLHeli ESC temperature is supported but currentely does not show right in the goggles even if IMO telemetry data is correct, probably needs a workaround, finally fltmode is displayed only is FAILSAFE is triggerd for the only matching flightmode between DJI and ardupilot is failsafe so the panel is hidden unless in failsafe where it shows !FS!, please remember that we do not drive the DJI framebuffer, we send telemetry and the goggles render it.
Right now Ardupilot supports all of DJI FPV Goggles V1.

Hwurzburg, 2021-04-18:
@yaapu Alex, I am confused, do we need to change the wiki in some way?

Target0815, 2021-04-19:
I guess only the line with "Distance Traveled" has to be removed ...

yaapu, 2021-04-19:
Sorry Henry, yes "Distance Traveled" has to be removed from the supported items

Hwurzburg, 2021-04-20:
done..closing

closed 2021-04-20 by Hwurzburg