ArduPilot/ardupilot_wiki #3438: Add AP_Periph parameters into Wiki somewhere

peterbarker, 2021-04-10:
We currently generate for all other vehicle types - we really should add AP_Periph

Hwurzburg, 2021-04-10:
@peterbarker @tridge not sure how this is actually going to be done for the wiki....aren't these just the same params as whatever library is being included? UAVCAN/GPS etc... and dont they vary by build for each device? isnt it more important that the GCS get the descriptions/metedata somehow to show in the SLCAN dialogs?

peterbarker, 2021-04-11:
AP_Periph is really just another vehicle type.  I think we should move it up, out of the `Tools` directory.

I think you found its parameter list here:
https://github.com/ardupilot/ardupilot/blob/master/Tools/AP_Periph/Parameters.cpp#L43

This PR: https://github.com/ArduPilot/ardupilot/pull/17138 will cause AP_Periph to appear here: https://autotest.ardupilot.org/Parameters/ as an AP_Periph subdirectory.

Hwurzburg, 2021-04-11:
the question is will that repeat all the #def libraries params also? I  imagine so....guess we include #ifdef'd out Plane params in the main Plane list also...guess, I am okay with all the re-duplication since it will be linked as a Adanced Configuration page in the AP_Pheriph eventual vehicle site...

brunoolivieri, 2021-04-11:
Parameter versioning does not download files from autotest, as far as I remember. Should I add and test AP_Periph as a versioned vehicle?

Hwurzburg, 2021-04-11:
I think we should go ahead and prepare to move AP_Periph up to a vehicle....we are about to add the autogenerated params into the dev wiki now, but could easily move that pages location to a new vehicle site when ready....@peterbarker....cant we move the wiki independently of how GITHUB gets re-org'd, when it does? we (Bruno?) can create a wiki site for AP_Pheriph now and I can start moving docs into along with the param page...it can initially be blank with an index.rst that just says under construction

peterbarker, 2021-04-11:
On Sun, 11 Apr 2021, Bruno Olivieri wrote:

> Parameter versioning does not download files from autotest, as far as I remember. Should I add and test AP_Periph as a versioned vehicle?

I think so.

peterbarker, 2021-04-11:
On Sun, 11 Apr 2021, Henry Wurzburg wrote:

> I think we should go ahead and prepare to move AP_Periph up to a vehicle....we are about to add the autogenerated params into the dev wiki now, but could easily move that pages location to a new vehicle site when ***@***.*** we move the wiki independently of how GITHUB gets re-org'd, when it
> does? we (Bruno?) can create a wiki site for AP_Pheriph now and I can start moving docs into along with the param page...it can initially be blank with an index.rst that just says under construction

Let's make sure everyone's good with that change.  Its possible for users 
people won't want it outside "Dev" (I would, but....)

The Wiki can be reorg'd easily.  There will be scripting things requiring 
changes - but they'll be very straight forward.

The really tricky bit will be making sure the existing links don't break - 
perhaps we'll need to work out all of the current pages and add a redirect 
in?

Hwurzburg, 2021-04-12:
PR posted....just need some more params emitted and will merge

Hwurzburg, 2021-04-21:
done...

closed 2021-04-21 by Hwurzburg