ArduPilot/ardupilot_wiki #3442: Text snippets for 4.1 features

andyp1per, 2021-04-10:
Gyro rates:

In prior ardupilot releases gyros and accels were either sampled at a fixed backend rate (generally 1Khz) or sampled at a higher rate and then averaged down to the backend rate, for instance sampled at 8Khz and averaged to 1Khz (so called fast sampling controlled by INS_FAST_SAMPLE). The "backend" rate is the rate at which gyro filtering takes place, i.e. low-pass and notch filters, and on smaller copters 1Khz is not high enough to avoid aliasing from noise at higher frequencies. For example a 3" copter might have a full throttle motor frequency of 600Hz. Since filters can only apply at half the sample frequency (the Nyquist limit)it is not possible to filter 600Hz noise when sampling at 1Khz and the noise will be aliased down to lower frequencies.

In 4.1 the backend rate has been made configurable for IMUs that support fast sampling (i.e. Invensense sensors) so that higher rates can be supported. The backend rate can by configured by setting INS_GYRO_RATE. The default 0 gives the same behaviour as 4.0. A rate of N gives a backend rate of 2^N Khz, so for instance a value of 1 gives a backend rate of 2Khz etc.

Note that this raises the rate at which all gyro filters are run and can be computationally expensive depending on the number of notches configured. Values above 0 are only recommend on F7 or H7 platforms and values above 1 should only be used with careful tuning.

Hwurzburg, 2021-04-11:
how does one know what is the limit for the higher backend rate? I think we would need a table vs sensor device name...and there must be some kind of absolute limit also that F7/H7 can cope with?

andyp1per, 2021-04-11:
It depends. The limiting factor is the number of notches. So if you are running double-notch-per-motor on FC's with multiple IMUs then the limit is going to be much lower than single notch per motor on a single IMU

Hwurzburg, 2021-04-11:
we have to provide some kind of guidance or this is not useable, otherwise we put this as a DEV only type of thing somewhere in the DEV wiki

andyp1per, 2021-04-11:
I would say the guidance is use values of 0 or 1. I don't think anything else is particularly wise or useful at the moment. 
You only really want 1 on smaller copters with F7 or greater.

Hwurzburg, 2021-04-11:
and with which sensors? and what if you use 1 with a F405 or sensor that cant support it? does it fall back to the lower rate?

andyp1per, 2021-04-11:
it will only work on sensors for which fast sampling works. It will be ignored if fast sampling is not enabled or the sensor doesn't support fast sampling.

Hwurzburg, 2021-04-11:
so why not just have it default to 1 and not talk about it...it will work fine when it can, and we just make a note in FFT stuff that if you are using a small copter with sensors that do not support fast sampling, you may get aliasing?

andyp1per, 2021-04-11:
It does default to 1 on F7 and H7

Hwurzburg, 2021-04-11:
okay, then this goes should go into a dev section...the average user need not know about this....

Hwurzburg, 2021-04-18:
#3463

closed 2021-04-18 by Hwurzburg